manuelbl/zy12pdn-oss #1: Question- Getting info for the sorce capabilitys

thetiana, 2021-03-11:
Hello, 

Is it possible to read the information about the source capability and how to send it to UART?

I am working around project to power my old laptop with USB-PD.
My idea is to use part of debug to read info about selected PD profile and mainly how much W can draw from the source. with the second MCU i will simulate DS2505 which is a communication chip inside the charger containing info about power capability of the charger. with this configuration i never harm the PD charger.

I look the debug code, but i cant find any code providing data how much amperes A can provide the charger, i think the voltage can be easily read and send to UART,  when V and A are available the W can be calculated and sent to UART

manuelbl, 2021-03-11:
This can be achieved with more or less the following steps:

- Ensure debug output is not activated (by not defining the `PD_DEBUG` macro)
- Change the code in `pd_debug.cpp` such that the UART code is always included
- Modify [`on_source_caps_changed`](https://github.com/manuelbl/zy12pdn-oss/blob/master/src/main.cpp#L141) so it writes the desired information to UART using `uart_transmit` or `uart_print`. The information is available on the `power_sink` object, in particular in the `source_caps` array.

Note that [`on_source_caps_changed`](https://github.com/manuelbl/zy12pdn-oss/blob/master/src/main.cpp#L141) must immediately select a voltage. Otherwise it is likely that you run into a timeout of the USB PD charger and it will reset. Just select 5V (`power_sink.request_power(5000);`) until your second MCU has decided for another voltage.

If the ZYPDN12 is also supposed to receive commands via UART, you will have to extend the UART code in  `pd_debug.cpp`.

UART TX is on pin PA2, UART RX is on pin PA3.

noahwilliamsson, 2021-03-16:
> If the ZYPDN12 is also supposed to receive commands via UART, you will have to extend the UART code in pd_debug.cpp.
> UART TX is on pin PA2, UART RX is on pin PA3.

FWIW, the first commit on my [apple-vdm](https://github.com/noahwilliamsson/zy12pdn-oss) branch enables both RX and TX and adds a stub for an UART debug loop.

Here's a [STM32F030F4 quickref](https://hw-by-design.blogspot.com/2018/07/arm-stm32f030f4-quick-reference.html) with pinouts.  I soldered a pair of enamel wires onto PA2 and PA3 those after removing the screw terminal on my board.

[image]

@manuelbl btw, really nice work on putting this OSS firmware (along with the detailed docs) together!

manuelbl, 2021-03-17:
Thanks for the update on your project. Looks very promising.