manuelbl/zy12pdn-oss #3: About those  unknown component

ID932569, 2021-11-27:
Hi, I did some search about those components, and here are the results

- U3 ME6203 Series  **ME6203A33M3G**  LDO from MICRONE
[https://item.szlcsc.com/89020.html](url)
- D1, D2 **CESD3V3D5** Uni-direction ESD Protection Diode
[https://item.szlcsc.com/92142.html](url)
- D3 **RB521S-40** Schottky Diode
[https://item.szlcsc.com/78476.html](url)

The following components are not unique

- SW1 **SKRPACE010** TACT SWITCH
[https://item.szlcsc.com/471390.html](url)
- LED1  
[https://item.szlcsc.com/347869.html](url)
- USBC1 USB Type-C Connector 16 Pin from Krconn
[http://www.krconn.com/content/149.html](url)

First time using Github, hope this will be helpful to you.

manuelbl, 2021-11-27:
Thanks a lot. I've updated the web page (incl. the schematic) with your information.

2212007054, 2021-12-06:
By the way ,now we can get the 3d model from grabcad.com
[https://grabcad.com/library/usb-c-pd-board-zy12pdn-1](url)

closed 2022-04-16 by 2212007054