manuelbl/zy12pdn-oss #5: PPS dynamic voltage support

Leseratte10, 2022-09-19:
Does this firmware support PPS with dynamic voltages, or can support for that be added? 

A Google search for the chip component number (FUSB302BMPX) brought me to [another repo](https://github.com/Ralim/usb-pd) that claims PPS would be supported. 

Looking at the source code of this firmware, II found code in pd_sink.cpp that looks like it could be supported: 
```cpp
        } else if (type == pd_supply_type::pps) {
            if ((capability & (3 << 28)) != 0)
                continue;

            max_current = (capability & 0x007f) * 50;
            min_voltage = ((capability >> 8) & 0x00ff) * 100;
            voltage = ((capability >> 17) & 0x00ff) * 100;
        }
```

but then, down below in the same file, there seems to be code that excludes all PPS modes when requesting voltage: 

```cpp
void pd_sink::request_power(int voltage, int max_current)
{
    // Lookup object position by voltage
    int obj_pos = -1;
    for (int i = 0; i < num_source_caps; i++) {
        auto cap = source_caps + i;
        if (cap->supply_type != pd_supply_type::pps && voltage <= cap->voltage && voltage >= cap->min_voltage) {
            obj_pos = cap->obj_pos;
            if (max_current == 0)
                max_current = source_caps[i].max_current;
        }
    }

    if (obj_pos == -1) {
        DEBUG_LOG("Unsupported voltage requested", 0);
        return; // no match
    }

```

Is PPS only partially implemented and is not going to work? Or is it supposed to work and I'm mis-reading the code?

manuelbl, 2022-09-20:
PPS is not fully implemented. There probably isn't a lot that's missing. But I can't tell when I will find time implement and test it.

manuelbl, 2022-09-26:
I've added support for PPS:

- `pd_sink::request_power(int, int)` requests power from a matching fixed supply or PPS capability.
- For special cases, `pd_sink::request_power_from_capability(int, int, in)` can request power from a specific capability.
- If a PPS capability is active, it is automatically re-requested every 8s. This is required by the standard as the power supply will otherwise reset (after 10 - 20s).

With a ZY12PDN trigger board, the firmware behavior does not change. The single button switches between the fixed supply capabilities.
 
To use PPS capabilities, a modified user interface is needed. A single button is likely insufficient to select voltage and current from a PPS capability in 20mV and 50mA steps, respectively.

Leseratte10, 2022-09-27:
Oh, cool, didn't expect this to be implemented so fast. Thanks!

I'm going to try to flash that to the module and see if it works. I was intending to mod the firmware anyways to ignore the button and hardcode a given voltage / current - I'm trying to convert an existing device with a stupid proprietary power plug to use USB-C instead, so that board would be built into that device with no need to ever use the button to change anything.

manuelbl, 2023-01-08:
There is now a wiki page with simple example for [modifying the firmware](https://github.com/manuelbl/zy12pdn-oss/wiki/Modify-the-firmware).

closed 2023-01-08 by manuelbl