manuelbl/zy12pdn-oss #10: Use as a USB-PD source?

fmckeogh, 2023-03-28:
Is it possible to use this hardware as a source and if so what software changes would be needed to support it?

To avoid an XY-problem, I have a solar USB-PD controller that advertises 20V even when the solar cells cannot sustain that. I can't really modify the solar controller, so I'm trying to make an adapter (made of two ZY12PDNs) which only advertises up to 15V then passes through subsequent communication to the solar controller.

manuelbl, 2023-03-28:
The hardware is suitable to implement a power source.

The software changes are not trivial as a power source is more complex than a sink. The current software implements the basic communication with the FUSB302 chip and the single message and the communication patterns of a sink. In order to add the messages and patterns of a source, a good understanding of the USB PD protocol is required. 

Also note that VBUS is not connected to the FUSB302 chip. So the VBUS voltage cannot be monitored. This could be a problem in your setup. 

In short: doable but involved.

fmckeogh, 2023-03-28:
Understood, thanks for your time!

closed 2023-03-28 by fmckeogh