manuelbl/zy12pdn-oss #13: Change voltage via esp?

WiredLife, 2023-09-30:
Hey, is its possible to trigger a specific voltage via an esp8266?
This would be very nice

manuelbl, 2023-09-30:
In order to control this board from an ESP8266, you would need to:

- Solder wires to unused pins of the microcontroller. PA0 thru PA4 and PB1 are available.
- Extend to software to act on commands from the ESP8266.

If you build the software with debugging enabled, PA2 is used as a serial output. So that's a start.

Soldering the wires isn't super easy as the chip has a small pitch size. It requires advanced soldering skills.

LRagji, 2024-01-08:
Maybe you can try [this](https://github.com/ryan-ma/PD_Micro) library and USBC PHY over I2C.. if anyone tries do post  results..