manuelbl/zy12pdn-oss #16: Any reason not following standard C++ capital letter classes naming?

diego351, 2025-04-24:
I'm not an C++ expert but it's hard for me to read the code not following C style naming. Is there any specific reason for the current state of being?

Thanks

manuelbl, 2025-04-24:
There isn't a single C++ naming standard. There are multiple ones.

At the time, I was working a lot with ESP-IDF and STM32 HAL. Both use naming a convention with underscores in type names.

diego351, 2025-04-24:
Are you sure your not confusing it with C? Can you give some reference to these C++ snake_case types naming _standard_?

manuelbl, 2025-04-25:
The C++ standard library is an example for C++ code using snake casing.

diego351, 2025-04-25:
That's all right then, thanks for educating me :). Sorry to bother!

closed 2025-04-25 by diego351