manuelbl/zy12pdn-oss #17: Does PPS support EPR voltage ranges?

diego351, 2025-05-19:
Hello, I really appreciate your work here. I needed proper PPS support and as soon as I got the proper supported board it miraculously worked.

Firmware-wise my question is whether I can ask for a ex. 28V using this software?

Thank you!

manuelbl, 2025-05-19:
28V is only available with the USB PD version 3.1. Currently, the software supports version 2.0. I'm  not sure if the USB PD chip on the board (FUSB302B) is capable of version 3.1. I will have to investigate it.

diego351, 2025-05-19:
I made some research and actually, the 20 - 48V PPS is called AVS and had been introduced with PD 3.2 specs, but just supporting EPR would be great.

One more thing, does an api allows to manage behavior depending of what usb cable is used using emarkers? Is it anywhere within any of abstraction layers?

manuelbl, 2025-05-20:
This library does not implement talking to the eMarker chip. I'm not even sure if the power sink (as opposed to the power source) is allowed to talk to it.

Probably, you can derive the certain capabilities from the advertised source capabilities. A maximum current of over 3A will indicate that a suitable cable with eMarker chip has been connected. The maximum current can be found in the _source_capability_ struct (see https://github.com/manuelbl/zy12pdn-oss/blob/master/include/pd_sink.h#L44).